motionlayer/Tinymovr #142: One shot command and feedback for high rate control

ThotAlion, 2021-05-07:
For high rate motor control at server level, it should be interesting to have a function in API which sends goal position by CAN and recup the feedbacks (position, speed, temperature, etc).
In "one shot" of CAN, you can give an order and get feedback.

Kind of :

...
[pos,speed,iq,temp] = tm.send_get_position(position,speed_feed_forward,current_feed_forward)
...

yconst, 2021-05-10:
Sounds good. The cleanest way this could be implemented is to 'overload' the read endpoints to accept setting values, by clearing the RTR bit. Then, the 'request' frame would also set the values, and Tinymovr would respond with a response frame as usual.
This could be implemented for a handful of real-time endpoints.
The data returned would be the same as the ones supported by the endpoint.

yconst, 2021-09-24:
Fixed by #155

closed 2021-09-24 by yconst