motionlayer/Tinymovr #188: Quadrature encoder support

gsion1, 2022-06-27:
Hi, 
Do you plan to add quadrature encoder support with A and B channel, plus index ? It seems that there is enough pins in the new tinymovr R5. It's sometimes hard and not reliable to use the board's encoder, especially in hard environnement or with outrunners motors (bracket can vibrate, bend etc)
The big issue with that is that the motor doesn't exactly know where is it when it starts and the motor may jerk, but maybe that's okay in some applications ?

Thanks a lot, have a nice day !

gsion1, 2022-06-28:
Hi, 
Thanks a lot for your answer 
It means that you plan to implement external encoder support ? I'm not able to see the roadmap, is it public ? 
Thank a lot !

yconst, 2022-06-28:
Yes we do have Hall effect sensor support in place, although we're still working on documentation. Next one is SPI sensor support. 

There is no public roadmap atm.

gsion1, 2022-06-29:
Thanks for your help ! Have a nice day

closed 2022-06-29 by gsion1