motionlayer/Tinymovr #193: Voltage increase stops motor functionality

amcunio, 2022-08-02:
We have been trying to increase the voltage slightly when running the motor controller. Pushing it from 32V->33V causes the motor to stop functioning. If it is underway this makes it stop and if attempting to calibrate this causes the resistance to be out of range.

Looking at the [resistance calibration sequence](https://github.com/tinymovr/Tinymovr/blob/9462d55c24cedc9aebed433fbcad0d7c9db48fa6/firmware/src/motor/calibration.c#L33) that fails using a debugger, observing the variables, the resistance @32V is R=~0.5 and when increasing by 1V, @34V R=~20. This is a sharp increase and I am not sure why this seems to be occurring.

Is anyone familiar with what in the firmware may be causing this to occur?

yconst, 2022-08-13:
Hi, I've tried this using a tinymovr R5, and it seems to be working without any issues, changing the voltage while in closed loop control does not affect functionality. I'm wondering if there could be some voltage transient affecting your case.

amcunio, 2022-08-22:
Thanks for looking into this, looking further it appears a voltage issue on our side.

closed 2022-08-22 by amcunio