motionlayer/Tinymovr #202: Current limit feedback

amcunio, 2022-08-22:
Hi,
We are wanting a way to get feedback back from the motor based on if the current limit is met. The motor will clamp the current setpoint being sent to the motor based on a settable current limit. As it stands an error is returned if the current level is too high (1.5 * this limit). We, however, would like some feedback on whether the limit has been reached and clamping occurs.

Detecting this shouldn't be hard, however we were wanting some feedback on what your preferred method of providing information to the user would be. Some possibilities we have are:
 - giving feedback in the get_vel_setpoint function if the velocity setpoint is causing the current limit to be exceeded
 - having a separate warnings channel (as currently only errors are returned and warnings can also be provided to the user)

yconst, 2022-08-23:
This issue contains very good ideas that I'd like to implement. The warnings channel, especially, is something that we've been considering ourselves. We are currently working on a [rewrite of the comms layer](https://github.com/tinymovr/Tinymovr/pull/201), which will allow gaining better insight in the state of Tinymovr. An endpoint for issuing warnings could be added there. The way we currently work with errors is that each submodule (e.g. encoder, controller, adc) has a bitmask for errors that corresponds to an endpoint. Similarly a bitmask for warnings could be implemented.

yconst, 2022-10-06:
We now have initial warnings endpoint support in the develop branch

closed 2022-10-06 by yconst