motionlayer/Tinymovr #203: Hall Mode

eufrizz, 2022-08-31:
I've been working on getting hall sensor feedback working (there were multiple bugs with selecting the different encoder type and calibrating it - the code seems pretty much untested with this mode). I have gotten it to a stage where it calibrates reliably and gets good feedback, so I can see the hall sector is picked up on and the position estimate increases/decreases as expected when you spin the motor by hand. 
Unfortunately however, the control loop seems unable to spin the actual motor (in velocity or current control). It will ramp up the current and make the motor stiff but it won't spin. Any ideas why? If you want to test it yourself, the working software is under the fixHallMode branch in my fork.
Thanks!

yconst, 2022-08-31:
Hi Eugene,

Thanks for raising these issues. Admittedly, the sector_map endpoint was never used much, but the rest of the encoder configuration and calibration process has been tested extensively before release. Nonetheless, I looked at the documentation following the issue you raised, and noticed some omissions. These to be addressed in a separate issue. 

Now to your problem. Most probably the gains need updating, like you correctly mentioned this is due to the change in mechanical resolution. I will update the docs with indicative values for the motor we tested on (it is a hoverboard motor), and add the necessary steps.

yconst, 2022-09-02:
#204 improved documentation regarding hall effect sensor, further information to be added.

yconst, 2022-09-07:
Work on docs is complete on this one. If you plan to create a PR we'll gladly merge it.

eufrizz, 2022-09-14:
Hi Yannis, thanks for updating the documentation, it was helpful! I've been testing hall mode on a MA330 encoder, which can simulate UVW output. I've added some convenience functions for reading/writing to the encoder registers (which are not actually used in the PR but could be useful for others in future), a new calibration function for the hall encoders, and tweaked some things in the CAN endpoints.
https://github.com/tinymovr/Tinymovr/pull/205

closed 2022-10-02 by yconst