motionlayer/Tinymovr #253: ModuleNotFoundError: No module named 'PySide2'

ator1811, 2023-04-10:
I installed the

yconst, 2023-04-10:
Please note the the guide and Studio release have just been updated to address this issue. So doing:

    pip install 'tinymovr[GUI]'
    tinymovr

Should install Studio 1.2.8 and should work. The only platform we're seeing issues is Apple ARM. If you are in this platform, we'll publish a new fix soon.

ator1811, 2023-04-10:
Thx, I'm on an Apple ARM. I followed the Anaconda procedure and not the PIP procedure.

yconst, 2023-04-10:
Note that you should still be able to use the CLI:

    tinymovr_cli

ator1811, 2023-04-13:
Thx, in the documentation under Anaconda installation [https://tinymovr.readthedocs.io/en/latest/studio/installation.html#anaconda-installation](url), the pip install command must be changed as well

closed 2023-04-17 by yconst