motionlayer/Tinymovr #302: R5.1 Hall sensor input maximum specification

pipipipi2002, 2023-10-04:
May I know what is the maximum voltage each of the hall sensor can take. I tried searching for schematics in the github to check but failed to do so. 

My motor takes in 3.5V-24V for its hall sensor and im thinking of supplying the hall sensor to the same voltage the motor is running on (12V nominal). I wanted to make sure that this is safe for the onboard hall sensor input.

yconst, 2023-10-04:
TL;DR: You should be ok, but 12V is not officially supported. The officially supported voltages for the hall sensor inputs are 3.3V and 5V.

Tinymovr interfaces to the hall sensor via an RC filter with in line 10k resistors. According to the [PAC5527 datasheet](https://www.qorvo.com/products/p/PAC5527#documents) (14.3.2 GPIO Current Injection), the device can tolerate a small amount of injected current for all IO pins, when the voltage is greater than 3.3V. This is stipulated as max 25mA per input and 50mA total. So considering the 3 hall inputs, we can deduct a max injection current of 16mA. 12V over 10k will give a bit more than 1mA, so you should be ok. However we have not attempted a 12V input, and we do not officially support this.

yconst, 2023-10-04:
Converting this to a discussion