motionlayer/Tinymovr #315: Motor configurations being re written during calibration

norsechurros, 2024-01-21:
I have these maxon motors that i am trying to configure with the tinymovr R5.1 , but the values get rewritten during call
bration. I have attached the datasheet with the highlighted values. I read a similar discussion with regards to a gimbal motor, but my motor is a high current motor that i intend to use on a grounded bot. 

[EN-21-306.pdf](https://github.com/tinymovr/Tinymovr/files/14000425/EN-21-306.pdf)

The phase resistance error is thrown as a result of which the hall sensors onboard the motor are not being calibrated. I have tried applying a bunch of different values for the phase resistances and inductances as well as the calibration current ,none of which seem to work. However, whenever i do run the calibration command the motor seems to move and draw a little bit of current ending with a beep sound, but the tinymovr goes into an error state(errors: flags.PHASE_RESISTANCE_OUT_OF_RANGE
).
How do i go about fixing this? 

[image]

yconst, 2024-01-21:
A resistance of 0.44Ohms should be accepted by Rx.x firmware, only M5.x firmware rejects resistances lower than 1.0Ohms. Can you post your `tm1. hw_revision` output?

norsechurros, 2024-01-23:
yes sure

[image]

yconst, 2024-01-23:
Ok so this is an older fw version where the hardware revision attribute was not supported. I'd suggest upgrading to the latest firmware, the instructions can be found here: https://tinymovr.readthedocs.io/en/latest/upgrade/upgrade.html#upgrade-using-activeflashlight-and-uart