motionlayer/Tinymovr #334: TinyMovr R5.2 drawing excessive current during change in modes

norsechurros, 2024-04-06:
These are my configs for a out of the box tinymovr:

[image]

When i now try setting the tm in velocity mode, calibration or position mode , it draws a very high current spike , and then doesnt end up changing its mode. This particular tinymovr was working for sometime under similar params before as well, until once when it kept drawing a lot of current and ended up burning a really costly motor. At the instance it was powered by a 6s 24V battery through a PDB. Now im trying to test it using a dc regulated power supply which limits current if it exceeds the limit. Why is this happening and whats the fix for it?

The specs are: 
Tinymovr R5.2 with out of the box firmware and studio 1.6.5
Maxon Motors EN-21-306
params are stated in the datasheet have been put in as configs here

I feel theres an internal problem with the tm in which case i want to first rule out other possibilities and then look for replacement.

yconst, 2024-04-07:
I'm sorry that your motor got damaged.

To be honest, I'm not sure what could cause this. Was it drawing current continuously when the damage happened? Or was it a current spike? 

From your settings, the only thing I can notice is the very high encoder bandwidth of 1500 Hz. For a hall sensored motor, with a low tick resolution, this should be way lower, usually less than 100 hz. Combined with the high gains, this could be a cause for current spikes. 

I'm leaving this here in case additional information comes up.

norsechurros, 2024-04-08:
The same bandwidth I had applied to all tinymovrs and motors i have worked with , and still do. Such a problem hadnt come up before, and still doesnt. I changed the bandwidth to a lower one and then tried callibrating it as well as trying to put it in position and velocity mode, all of which results in the same high spike in current draw.

Its an internal problem of this tm.

yconst, 2024-04-10:
Got it. Did not understand that this concerned only a single unit. As this is the case, I'm moving this to the discussions.

I do not have an answer as to why this issue started appearing. If I would hazard a guess, could be either damage to the gate driver, leaving the mosfet on, or an issue with the ADC reporting incorrect current values, thus allowing the current to exceed the regulated value. But these are just hand waving.