motionlayer/Tinymovr #371: Unable to use CLI/python api to set external encoder

omelia-iliffe, 2024-09-18:
Hi, 
I'm trying to get my TinyMovr M5.1 setup with an external encoder (AS5047) but i'm running into serveral issues.

I first installed the studio using python3.12. The studio opens but when I go to select the encoder only Internal and Hal are presented as options. 
I then tried following the Example given in the docs to [External AS5047 Sensors](https://tinymovr.readthedocs.io/en/latest/sensors/sensors.html#external-as5047-sensor-for-commutation-and-positioning) which I assumed was instructing me to use the python api.
I copied the example code from the [Integrating page](https://tinymovr.readthedocs.io/en/latest/protocol/integrating.html)
```
import can
from tinymovr.bus_router import init_router
from tinymovr.config import get_bus_config, create_device

params = get_bus_config(["canine", "slcan_disco"], bitrate=1000000)
init_router(can.Bus, params)
tm = create_device(node_id=1)

tm.controller.calibrate()
```
This code fails immediatly as the init_router function requires a logger parameter. After searching for ages to find out what that parameter should be (Hint it should be pythons built in logger initialized with the `configure_logging` function) I then tried to use the code from the AS5047 example to setup the external encoder and calibrate.
```
tm.sensors.setup.external_spi.type = tm.sensors.setup.external_spi.type.AS5047
tm.sensors.select.commutation_sensor.connection = tm.sensors.select.commutation_sensor.connection.EXTERNAL_SPI
tm.sensors.select.position_sensor.connection = tm.sensors.select.position_sensor.connection.EXTERNAL_SPI
tm.calibrate()
```
But that doesn't work either. 
```
Traceback (most recent call last):
  File "c:\Users\Omelia\repos\tinymovr\calibrate.py", line 12, in <module>
    tmx.sensors.setup.external_spi.type = tmx.sensors.setup.external_spi.type.AS5047
                                          ^^^^^^^^^^^
  File "C:\Users\Omelia\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\avlos\definitions\remote_node.py", line 48, in __getattr__
    raise AttributeError(__name)
AttributeError: sensors
```

I then realised their was the tinymovr_cli that I hadn't tried but that wasn't working for me either.
It finds my tinymovr (tm1) but everything I type results in errors
```
In [1]: tm1

File ~\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\avlos\definitions\remote_attribute.py:62, in RemoteAttribute.str_dump(self)
     61 def str_dump(self):
---> 62     return "{0} [{1}]: {2:.6g}".format(
     63         self.name,
     64         self.dtype.nickname,
     65         self.get_value(),
     66     )

ValueError: Unknown format code 'g' for object of type 'str'
```

```
In[2]: tmx.sensors.setup.external_spi.type = tmx.sensors.setup.external_spi.type.AS5047
File ~\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\avlos\definitions\remote_node.py:48, in RemoteNode.__getattr__(self, _RemoteNode__name)
     46         return attr
     47 except KeyError:
---> 48     raise AttributeError(__name)

AttributeError: sensors
```

I also tried with python3.10 on a linux machine to see if it was python3.12/windows 11 that was causing the problems but no to avail.

Any help would be appreciated.

yconst, 2024-12-03:
Hi,

Apologies for the late reply. I'm looking into this and will reply to you very soon.

yconst, 2024-12-08:
We fixed the logger and object print issues in the latest Tinymovr Studio release (2.3.2):
https://pypi.org/project/tinymovr/

please use `pip install --upgrade tinymovr` 

We also improved the documentation to refer to `tm1` insteda of `tmx` with a note for applying to other nodes in the bus.