motionlayer/Tinymovr #378: NO response by tinymovr studio on operating a bldc motor, tinymovr R5.3 with canine v1.4 adapter

TayyabAsim047, 2025-02-05:
Tinymovr Version: R5.3
BLDC Motor Model: [JD-power DCH-6829F flat dc hollow shaft brushless motor with switch hall sensor]
Bus Interface: [SLCAN]
Tinymovr Studio Version: [2.3.2]

Issue Description:
I am using a Tinymovr R5.3 board to control a BLDC motor via Tinymovr Studio, but I am not getting any response when sending commands. The board is detected, and communication over [CAN/SLCAN] is established successfully. However, when issuing movement commands, the motor does not move, and there is no feedback in Tinymovr Studio.

Expected Behavior:
The BLDC motor should rotate based on the commanded setpoint, and Tinymovr Studio should provide real-time feedback.

Observed Behavior:
No motor movement
No output or error messages in Tinymovr Studio
Board appears connected but does not execute commands

Troubleshooting Attempts:
Verified power and wiring connections
Confirmed motor parameters (resistance=1.6ohm, inductance=2.11mH, pole pairs=11)
Tested different bus interfaces (tinymovr_cli shows device, but no control)
Reinstalled Tinymovr Studio and drivers

Any guidance on debugging or further troubleshooting would be greatly appreciated. Thanks!

[image]
[image]

yconst, 2025-02-05:
Please post the motor parameters including errors.

Please note that Tinymovr R5.3 does not support motors with internal resistance > 1Ohm

yconst, 2025-02-05:
Please note that Tinymovr R5.3 does not support motors with internal resistance > 1Ohm

yconst, 2025-02-05:
Converting this to a discussion